Miq1/FredBoard #20: Flickering LEDs are back.

Smelter, 2024-10-28:
### Version

alpha5

### What is the issue?

These are LEDs on the Left Hand End under the 'C' label.
Looks like a retro step as all was well with 4g.

https://github.com/user-attachments/assets/8d77b9f2-49b6-4f30-bab1-952cadc8f248

amaskell999, 2024-11-04:
I think this has been fixed in alpha5.1.A

Miq1, 2024-11-05:
Does rather look like the DMA quirk is back Fred had earlier.

Frudah, 2024-11-05:
If you reset your settings on install of the new firmware I believe the default is 40 for noise floor. I can make the lower limit higher for roll over… or assign it to a sub menu so it allows up and down so there is no roll over. 5.1.a does fix it I believe. The buffer function is still the same. I plan on improving it. What I did find, was the strummer1 was always active and cycling and it was lagging down the led timing. 

I have now made it so the strummer isn’t activated until a button with strum is pressed and is deactivated after a press. I’m sure I can optimize the strummer a bit to not make it as resource intensive later.

Frudah, 2024-11-05:
… and closed

closed 2024-11-05 by Frudah

Frudah, 2024-11-07:
I also found that strummer 2 and strummer 3 modules were continually looking to make sure there were no unresolved strum notes to turn off, robbing the chip of thousands of clock cycles per midi pulse. 5.2 has this fixed and should further reduce any of these erroneous pulses being presented.