Miq1/FredBoard #21: LED splatter pattern for single key press.

Smelter, 2024-11-04:
### Version

alpha5.1

### What is the issue?

I have tried multiple sensitivity and noise threshold settings with no change.

https://github.com/user-attachments/assets/e55cc6d0-6709-494d-b4df-199bfa00ce4b

Miq1, 2024-11-04:
This is a feature Fred announced with alpha5. You can switch it off in the Settings.

closed 2024-11-04 by Miq1

Smelter, 2024-11-04:
I didn't expect it to be on by default. Looks like we need a user
preference save function :-)

Frudah, 2024-11-07:
it is there. close to bottom of settings... select "off". put it on by default so you'd see it and can disable it if you don't like it. make sure to use shutdown function to power off. this will save settings. settings are not saved unless "shutdown" is used... or it is saved and recalled from a user set.

Smelter, 2024-11-07:
All perfectly reasonable. And yes I get it now. I know we are still at the Alpha testing stage, but when something appears, and its set on at start up, my brain says "that's new, do I like it, can I turn it off and it stays off" Consider my ramblings as innocent and exploring. I have beta tested other software and heard back that no one else has spotted that, which has lead to  interesting developments. So these are not complaints just observations.