Miq1/FredBoard #23: Light Brightness : Left hand side only.

Smelter, 2024-11-04:
### Version

alpha5.1

### What is the issue?

Only the left hand side brightness is adjustable. Right hand side is locked to "LOW LOW". No menu entry to say if sides can have separate brightness.

https://github.com/user-attachments/assets/bc0a72e5-6b03-4a7a-9690-f55e7e78dd4b

amaskell999, 2024-11-04:
Not happening with mine, but I'm already on alpha5.1.a

Smelter, 2024-11-05:
> Not happening with mine, but I'm already on alpha5.1.a, if that makes any difference?

It's 5.1.A that I am using! Fred is getting versions out quicker than the "version" option here gets updated.

Miq1, 2024-11-05:
> It's 5.1.A that I am using! Fred is getting versions out quicker than the "version" option here gets updated.
Aw, my fault. Fixed in a second.

Frudah, 2024-11-05:
Alphabetic subversions usually contain minor patches and it is unlikely that a lettered subversion will act significantly different than a numerical upgrade. Have you reset your settings in the settings menu. Can’t think of how that may affect your issue, but is typically required on most upgrades as I get things settled. Once we get to beta this will no longer be necessary as all options and menus will be finished.

Frudah, 2024-11-07:
one temporary note... and I am going to close... make sure you are hitting the enter key on the mode you have selected. if you boot up and it shows an x on the scales or classic... this has been saved during shutdown. even though it shows an x, it has not fully loaded the mode options. click it again and you should see the brightness levels correctly changing now. in 5.2 I have ensured all the grid options load for the saved/recalled mode. for now, just click it again.

closed 2024-11-07 by Frudah