Miq1/FredBoard #24: Keypress indicator lights appearing in wrong place on right keypad.

amaskell999, 2024-11-04:
### Version

alpha5.1

### What is the issue?

Includes alpha5.1.A.
When pressing keys on the right keyboard, any keys from columns 5-8 cause the keypress indicator to light white on the top row column 5 instead of the actual key pressed. The same happens for columns 9-12 when the top row column 9 key lights. The erroneous light goes off if either of the top row keys are pressed but it doesn't fix the fault. The MIDI output of the key presses is not affected.

Frudah, 2024-11-07:
which mode is this in? scales? classic? both? others?

Frudah, 2024-11-07:
nvm. I have reproduced this issue on my board. guess you don't like the exploder :P 
I'll have it worked out for 5.2

Frudah, 2024-11-07:
marking closed. erroneous code has been identified and changed for 5.2

closed 2024-11-07 by Frudah