Miq1/FredBoard #32: No Shut Down 6v0

Smelter, 2024-11-19:
### Version

alpha5.4

### What is the issue?

Alpha 6 : Shutdown menu keypress has no action. 
Are the settings saved?

Menu screen will blank out after a few seconds. Missed that on the recording.

Smelter, 2024-11-25:
Progress comment from developer in 'Messenger Chat' :- "it does save before whatever it doesn't do"
So presumably its a screen menu glitch and is safe to power down after execution.

Miq1, 2025-01-07:
Pushed up to "Bug", since settings are not saved any more with 6.6.

Miq1, 2025-02-01:
'System->Save All' works now (Stechapfel.6.6.Fa)

'System->Shutdown' is bringing the board into some limbo - middle keys and display remain lit, as do the top row buttons.