Miq1/FredBoard #35: ABOUT could have unique build number...

distrakt, 2024-12-10:
### Version

alpha6

### What is the issue?

With the sometimes-rapid-iteration, it would be useful to see a completely unique build number in the About box.

(In my projects, every build runs a script that increments version.h, so the about box or --help shows things like "build 4037".)

Can also include __DATE__ and __TIME__ to get a very unique number.

Ideally the build number is also on the file name, like FIRMWARE_5.6_4037.333 or whatever. (Right now I have two quick iterations in my downloads which are distinguished by an extra period in the filename :) )

Frudah, 2024-12-14:
Which builds are those? I usually use numbers for significant changes…. And letters for minor fixes. I am pretty lazy about ensuring the about page gets upgraded consistently, mostly because the quick iterations, but typically ensure the file names are reflective.

Miq1, 2024-12-15:
@Frudah I think @distrakt is referring to an automatically generated number (by a script in the build process) that will be incrementing with every build. I have similar mechanisms in Platformio to maintain this number and generate a build version string from it. Once implemented you will not have to care about it any more.

Frudah, 2025-02-01:
While there are ways to implement this, I will continue applying manually. Versions will be coming slower than before, hopefully reducing the amount of manual error / version confusion. Versions will also have a 'major' version designator to differentiate between releases / sub-releases. 
Reopen if confusion persists :)

closed 2025-02-01 by Frudah