Miq1/FredBoard #36: Chord display flickering/invisible

Miq1, 2024-12-10:
### Version

alpha6.5F

### What is the issue?

The chord or note display screen (Main menu, left arrow button) is unreadable. There is a constant, but flickering "-2" left in the middle, while chords/notes only will shine up partly and corrupted.

@Smelter: you found it originally :wink

Miq1, 2024-12-14:
Update on version TallFescue6.5999: The "-2" is gone, but the display still is flickering and incomplete.

Smelter, 2024-12-14:
The -2 is gone but the user name it replaced has not returned. The user name is only seen on power up when the light show that refusers to be able to turned off runs.

Frudah, 2025-02-01:
still minor flickering / screen refresh, but IMHO, readable and functional.

closed 2025-02-01 by Frudah