Miq1/FredBoard #39: User Mode Key 0: fails to send

distrakt, 2024-12-18:
### Version

TallFescue.6.5999

### What is the issue?

in "User Mode" the top left key midi pitch=0, lights up but _**does not send a NOTE ON.**_ upon release, it does send a NOTE OFF.

Here is midi data from pressing keys 0 1 and 2:

```
FRED:FB: 90 01 04
FRED:FB: 80 01 00

FRED:FB: 80 00 00

FRED:FB: 90 01 0d
FRED:FB: 80 01 00

FRED:FB: 90 02 04
FRED:FB: 80 02 00

FRED:FB: 90 01 0a
FRED:FB: 80 01 00

FRED:FB: 80 00 00

FRED:FB: 80 00 00

```

Frudah, 2025-02-01:
this was a shortcut I used. any note less than 1 does not send. I will leave open and change the "dead note" indicator.

Frudah, 2025-02-13:
Repaired function for user mode to not ignore Midi Note Value 0x0. Fixed for 6.6.0Fc

closed 2025-02-13 by Frudah