Miq1/FredBoard #40: Hard strike can produce multiple midi events

distrakt, 2024-12-18:
### Version

TallFescue.6.5999

### What is the issue?

I did this in "USER MODE", but likely this happens in other modes too.

A hard fast strike (like a drumming tap) can produce multiple key on/key offs. The printout below includes time stamps in microseconds. We can see it's more likely to happen on hard strike (6b and higher) and if it's very brief. 

```
FRED:FB: 1642560339 90 05 6b
FRED:FB: 1642567488 80 05 00
FRED:FB: 1642571472 90 05 5d
FRED:FB: 1642744616 80 05 00

FRED:FB: 1648271776 90 16 05
FRED:FB: 1648504943 80 16 00

FRED:FB: 1664965023 90 19 73
FRED:FB: 1665178429 80 19 00

FRED:FB: 1671630723 90 19 75
FRED:FB: 1671679852 80 19 00

FRED:FB: 1674339034 90 19 77
FRED:FB: 1674390804 80 19 00

FRED:FB: 1676157987 90 19 77
FRED:FB: 1676161481 80 19 00
FRED:FB: 1676164242 90 19 46
FRED:FB: 1676204122 80 19 00
```

Frudah, 2024-12-21:
you may want to adjust the "Noise Floor" value in sensitivity. sounds like a cross talk issue fixable with adjusting the analog filter floor via the "Noise Floor" value being increased.

Miq1, 2024-12-21:
The remedy may work; it does not look like a cross talk issue, though, as it is always the same note triggered again. One possible reason could be the plastic contact sheet is slightly "bouncing" wenn hit hard enough, another would be an overflow of the velocity value when brought to the 0-127 range.

distrakt, 2024-12-21:
I'll experiment with the noise floor setting.
it does seem to be a "sometimes" thing. this morning I'm not seeing it happen. maybe i dont have enough static electricity from the day in me yet. I'll collect electrons during the day and try again.

Frudah, 2025-02-01:
should be fixed with new version 6.6Fa. please re-open if you find otherwise.

Frudah, 2025-02-01:
should be fixed with 6.6Fa. Unable to replicate in newest version. Please reopen if still occurring.

closed 2025-02-01 by Frudah