Miq1/FredBoard #41: Key brightness changes after 2nd attempt only

Miq1, 2024-12-23:
### Version

TallFescue.6.5999

### What is the issue?

When changing Key Brightness in the settings, the top row brightness will change immediately, but the main keyboards will need a second button press.

Smelter, 2024-12-23:
Fred recently introduced a shade difference between the main notes and the #/b notes. I personally do like it but some may prefer the option of "classic" levels of all the same.

Miq1, 2024-12-23:
> Fred recently introduced a shade difference between the main notes and the #/b notes. I personally do like it but some may prefer the option of "classic" levels of all the same.

I know that, but that is not the issue. You need to press the middle button twice when setting to make a brightness change visible.