Miq1/FredBoard #45: "5" key is a trapdoor into Pentatonic

Miq1, 2025-01-22:
### Version

TallFescue.6.5999

### What is the issue?

If pressing the "5" key once, you are trapped inside Pentatonic mode on the right hand keyboard. There is no other way out as the Scales menu.

I would expect the "5" key to toggle between Pentatonic and chromatic modes.

Side track: **if** you switched back to chromatic mode with the Scales menu, the "5" key will remain functionless until you pressed the "12" key at least once.

Miq1, 2025-01-24:
Fixed in Stechapfel6.6a, "12" brings you out of it.

Frudah, 2025-02-01:
need more info. The intention of the 5 key is to switch it permanently to pentatonic (no toggle).... with any of the modal keys, user slots... the '12' key .... or selecting a scale in the menu. 

it is intended for only the user slots / '12' key to toggle back.

Miq1, 2025-02-01:
Now (Stechapfel.6.6.Fa) at least the mode buttons will bring you back. 
The "12" button does change the layouts for Pentatonic mode using the same logic as in the "normal" scales - fine.
I will try to see the "5" button like another mode button for now.

closed 2025-02-01 by Miq1