Miq1/FredBoard #55: Better way to show chord notes in 'Live Update' mode

Miq1, 2025-02-01:
### Version

Stechapfel.6.6F

### What is the issue?

I would like to propose a different approach to the chord notes being shown on the right keyboard in the different layouts:

Given the chord Fmaj7 in octave 3 - notes are F3, A3, C4 and E4.

Let the note keys on the right light in white for a short period after the chord key is pressed:

[image]

The example is shown for the chromatic layout, but the principle will work in any layout.

This will help learning which keys are used in a chord, if the scales will remain at their root note position.

abdinfos, 2025-02-05:
Good idea but should not replace normal Live Update. Because of the isomorphic nature of the layout of the current Live Update keys layout which is also of interest in the first place. Could be a second Live Update mode.
By the way, why are the octave key in vertical layout on you drawing? It disturbs two octaves instead of one.

Miq1, 2025-02-05:
> By the way, why are the octave key in vertical layout on you drawing? It disturbs two octaves instead of one.

Fred has changed that a couple of times, faster than I could follow 😉 I do not know if he is finally decided now.

abdinfos, 2025-02-06:
On second thought, I see this more as a Chord Sense or Chord Show or whatever it's called, independent of Live Update. It could be enabled or disabled by setting.
It would come in very handy, whether for learning, guiding or even checking the operation of the chords generated by the left keyboard. For instance it could also be used to guide play by adding ephemeral notes to chords with the right hand, notes that you wouldn't have to add to chords with the center keys. Nice to vary your playing, adding a bass an octave lower while visualizing its position, guiding the melody, etc…
The pads concerned could perhaps flash as long as the notes played by the left keyboard are active (or as long as another left chord pad is pressed), or (and I think I prefer this) have a change of intensity (plus or minus).
What's more, this doesn't prevent you from keeping Live Update mode as it is now. On the other hand, if Live Update is not active, any unused keys should be used to complete the scales, making it easier to play chords, and should be taken into account by Chord Sense.

abdinfos, 2025-02-06:
> > By the way, why are the octave key in vertical layout on you drawing? It disturbs two octaves instead of one.
> 
> Fred has changed that a couple of times, faster than I could follow 😉 I do not know if he is finally decided now.

I thought the octave keys were only in the vertical position with the guitar display, when in fact they are string shift keys, which makes sense

Miq1, 2025-02-06:
> I thought the octave keys were only in the vertical position with the guitar display, when in fact they are string shift keys, which makes sense

In fact they were at one time, but Fred managed to render the former octave down key useless in most other layouts 🤣  
We will see what comes out at the end.

Frudah, 2025-02-13:
should be fixed now... (chord octave buttons). only the guitar layout is meant to be vertical...

will circle back on alternate live note update option.