Miq1/FredBoard #56: Light Board - access and use issues.

Smelter, 2025-02-02:
### Version

Stechapfel.6.6F

### What is the issue?

"Light Board" sub menu reads :
Mode : LIGHTBORD       - What are we going with here? with or without the letter A?
< Chromatic
All Lit >

Now the arrow entries in this menu are not consistent in response. I have been able to turn off all the LEDs in this menu and turn them back on.
Also I have been able to "X" and un"X" the top level menu listing.

The sub menu is time entry dependent, and returns to the same main menu position. (thank goodness, read on)
However for key entries 'O' '<' it exits to the rolling number. (can't remember what it is I read that is) press '>' and we are back to the sub menu with no time out. Press '>' at this screen and nothing happens, other than the number count is refreshed. Wait awhile and it times out back to the main menu but at the "Scales |>" entry, 8 steps up from the "Light Board |  X" entry, which if you press 'O' you are then locked into the sub menu where neither '<' or '>' works so you have to wait until the time out takes you back to the main menu.

And if that was not confusing enough. there is a surprise if you try to access the "Manual" sub menu, as you are back to the "Light Board" sub menu where pressing the 'O' key makes the sub menu title say "Scales" and further pressing of 'O' starts a scroll through colour pallet selections.

Looks like there are cross reference/links between menu entries and possible code re-use that it getting well confused.

Frudah, 2025-02-03:
Where is the button to give you the ‘Greatest bug tester award’?

Miq1, 2025-02-03:
Added it 😄

Smelter, 2025-02-03:
> Where is the button to give you the ‘Greatest bug tester award’?

Glad to be of service. Actually it's nice to use the methodical approach I used when employed, into my retirement. Just glad others know the music theory side for testing, that is what I bought the TB for, but coming without a manual and having issues, how was I to know how if it worked or even if it was doing it right?

closed 2025-02-13 by Frudah